ROS-Mobile/ROS-Mobile-Android #2: SSH Connection data not saved

l-schilling, 2020-05-26:
If you use the SSH tab and change the IP adress, username and password and then at some point reopen the app, the data is not saved. Instead the default ssh values are there.

NRottmann, 2020-06-04:
Thanks for your comment. We currently work on adding this feature into the app. We will keep you updated about the progress.

NRottmann, 2020-06-08:
Hi, SSH access data is now saved. You can download the current version of the application here:

https://github.com/ROS-Mobile/ROS-Mobile-Android/tree/Update_1.0.1

and follow up the installing instructions from here:

https://github.com/ROS-Mobile/ROS-Mobile-Android/tree/Update_1.0.1#built-from-source

Best, Nils

closed 2020-06-08 by NRottmann