ROS-Mobile/ROS-Mobile-Android #4: No Images sent to Mobile Device

NRottmann, 2020-08-07:
I did not receive images from my camera which is mounted on a jackal robot. Error:

2020-08-07 14:50:36.929 24054-24199/com.schneewittchen.rosandroid E/UpdatePublisherRunnable: java.lang.RuntimeException: java.net.UnknownHostException: Unable to resolve host "cpr-j100-0406": No address associated with hostname

Marcus-Davi, 2020-09-07:
Same problem here, however I think it's due to version

" Client [/Camera] wants topic /camera/image_raw to have datatype/md5sum [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f], but our version has [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743]. Dropping connection. "

I'm using noetic.

NRottmann, 2020-09-08:
We are trying to fix the problem and will integrate it into the next version.

NRottmann, 2020-09-14:
This problem is solved (check the MapView2 branch) and the solution will be integrated into the upcoming version.

closed 2020-09-14 by NRottmann