ROS-Mobile/ROS-Mobile-Android #5: Unable to connect to a given IP address

Marcus-Davi, 2020-08-24:
I've tried using both the GPlay app and the compiled .apk and failed to connect to a given IP address to rosmaster.

I'm currently testing the application in Android Studio and even though I'm not an expert Java programmer I identified that the IP is somewhat **hardcored** - meaning that no matter what I insert as the master IP, it always tries to connect to "192.168.0.0".  I've changed in the source code every entry of "192.168.0.0" to my local IP and it worked. (Tested joystick only publishing)

I've changed the emails in the files 

- ConfigModels.java
- ConfigDatabase.java

I believe that this issue happens with the SSH socket as well.

Marcus-Davi, 2020-08-24:
Doing more testes.. It seems like the UI is not changing the back-end variables.. Changing to joystick topic name won't change the actual published topic name in the master (checked with rostopic list in the PC)

williamsantosferreira, 2020-08-24:
Yes I'm having this issue too

LuanAmaral, 2020-08-24:
I'm having this problem too.

NRottmann, 2020-08-27:
Did you hit enter on your mobile device keyboard after inserting the master IP? That will store the IP address. You can check this by switching to another tab (e.g. ssh tab) and back to the master tab, If the IP is now correctly set, it should hopefully work.

Marcus-Davi, 2020-09-07:
The IP remains unchanged. I insert the roscore IP, press the keyboard enter, it refuses to connect. I then change tab to SSH, and go back to Master Tab. The Default IP reappears, indicating that it really didn't store the inserted IP!

Marcus-Davi, 2020-09-07:
It apparently worked when I pressed enter in the keyboard (Playstore app) ! The outline color (blue) around the text box even vanished as a visual indication that something changed.  However, a similar problem appears in the "Details" and "SSH" tabs, where I am given the option to choose the topic name for camera / joystick / IP address for ssh. When I press enter, the outline color does not disappear, and the cursor for text input remains. I've tried changing the topic name of the joystick and it does not change.

NRottmann, 2020-09-08:
For applying changes in the Detail Tab, each node has a circled arrow at the top right. If you changed something for a node (e.g. the topic name), just hit this button and all changes are applied and the nodes will be reconnected with the new specifications.

Marcus-Davi, 2020-09-10:
Ok, all text input seems working in the master version, but not the GPlay version, just tried it. In my Samsung S3 Tablet sometimes it doest not show all text, i.e. I use "ls" command and it shows only 4 of the 21 folders.

Also, while trying "rostopic echo /gps" to a publishing topic, I can't "Ctrol + C" it, haha (at least i don't know how), requiring me to reconnect. Maybe add a button ? Moreover, while listening to a NatSatFix topic, sometimes it "eats" the text, not showing the actual coordinates. Maybe due to the scroll down animation. It keeps showing "---", and sometimes it prints text over text.

blubbi321, 2020-09-28:
Who is maintaining the GPlay version? (wondering whether its worth to compile myself or wait for an update on GPlay)

Torellin, 2020-09-28:
I and also Mr. Rottmann are responsible for the Google Play version.
It depends. If you want the current stable version, download the app from Google Play. The master-branch also contains a stable-ish version but with new features und bug fixes.
We are currently working on bigger updates for GPlay, but this will take another week or two.

blubbi321, 2020-09-28:
regarding the stable version on the master branch you mean this one here https://github.com/ROS-Mobile/ROS-Mobile-Android/blob/master/app/release/app-release.apk ?

ok, if the next release is within weeks then i think waiting is the "quicker" option for me ;)

Torellin, 2020-09-28:
The apk-file in this repo is the same as the GPlay version. But I will build another apk for you in the next few days that you can download and install on your device.

NRottmann, 2020-11-05:
A new release version will be uploaded on Google Play within the next week. Two new issues have been separated out of this,

NRottmann, 2020-11-05:
A new release version will be uploaded on Google Play within the next week. Two new issues have been separated out of this, #18 und #19.

closed 2020-11-05 by NRottmann