ROS-Mobile/ROS-Mobile-Android #8: Camera drawing and Compressed image md5sum

Marcus-Davi, 2020-09-14:
Hello ! 

The camera is working nicely ! Subscribing to a publishing image topic has been successfully displayed in the app. I've encountered two issues

- The compressed image message seems to be still not updated for ROS Noetic, and not yet tested on previous ROS versions. The raw image worked just fine.
- Whenever changing the box dimensions in the DETAILS menu, sometimes the outline drawing box does not fit the camera image. Maybe it is related to the aspect ration, which has to be preserved ? 4 width x 3 height and multiples are OK, but anything else has weird behavior, like the camera image appearing in the middle of the VIZ menu and the outline box not involving the image at all (in case of  9 width x 2 height). I am aware that's quite an absurd image ratio, but I think it should be addressed somehow. Maybe forcing constant ratio for the camera dimensions or re-scaling the image.

All else its very nice !

Marcus-Davi, 2020-09-14:
**Update** 

The compressed image actually works. I have to specify the directly to the compressed topic (i.e /usb_cam/image_raw/compressed). It's not as adopted by the image_transport standard, which is to subscribe to the "general" (i.e /usb_cam/image_raw)  topic and automatically figure out the image_transport. Works for me anyway!

NRottmann, 2020-09-22:
We integrated improvements into the Update-1.0.2 branch. The following features are now available

- No update button required, update as soon as anything changed in the details
- Available topics with suitable message types are now shown in a spinner based on the chosen message type
- A debug node has beed added

Hope this solves some of the above problems

NRottmann, 2020-11-05:
With the update 1.1.0 all camera problems are solved. Automatic detection of correct msg types ensures functionality.

closed 2020-11-05 by NRottmann