ROS-Mobile/ROS-Mobile-Android #9: GPS Map draw without internet

Marcus-Davi, 2020-09-21:
Hello ! 

The APP is very stable and working properly. However, I often find myself connected to the mobile robot own network, and my Android is unable to access internet. This makes the app unable to draw the GPS location within the map. I wonder if is there any way I can download the offline map for a certain location ? How is it supposed to work without internet access ?

NRottmann, 2020-09-22:
For the GPS View, we use Open Street Maps (OSM). There are license issues if downloading maps. Thus, we only support online maps. For accessing the mobile web while being connected to the robot activate "Mobile data always active". You can find this in the developer options.

NRottmann, 2020-11-05:
Minor remark: Up to now make sure to activate "Mobile Data" after connecting with the ROS master, otherwise the application might not find the correct idea.

closed 2020-11-05 by NRottmann