ROS-Mobile/ROS-Mobile-Android #13: Only one widget of type works

Cycov, 2020-11-04:
### What is the issue:
Only the last widget works

### Where/When:
I added two joystick widgets and only the last modified works

### Why
In `RosRepository.java` when a change is made, that node with the default name is getting reregistered this.reregisterNode(node);` on line 265, therefore taking precedence over the others. This is due to the fact that only one node with a particular name can be active at any moment in time.

### Solution
1. Append a UUID to unnamed nodes's default name
2. Warn the user of the naming conflict

Torellin, 2020-11-05:
Hey, do the joysticks have equal names and/or are publishing to the same topic?

To my understanding you shouldnt be able to publish to one topic with more then one node.

Have you tested this for the latest version 1.1?
Greetings, Nico

Cycov, 2020-11-05:
I have not tested on the latest version as this issue was opened before the merge. Visually, the joysticks names are different, but in reality they are not. When I manually renamed them to something different I am able to publish on the same topic.

I redid the tests now and it seems that the problem does occur only on the play store version. Nevertheless, you can publish on the same topic on the previous version, didn't test on the new one.

I guess you can close the issue, but a warning when two nodes with the same name should be available.

closed 2020-11-05 by Cycov