ROS-Mobile/ROS-Mobile-Android #26: Unable to connect to master

SimoCerra, 2020-11-24:
I'm trying to connect to the master node running on my PC. Actually, I'm using my smartphone as Wi-fi hotspot and my PC is connected to it. I have writed the ip address of the master node in the specific field of the app but when I clicked on the "connect" button the app is not able to perform the action.
Have you any idea what is going wrong?

Torellin, 2020-11-24:
Hard to say. We have actually not tested the cell phone as a Wifi hotspot yet. But I will have a look at it.

Torellin, 2020-11-24:
Have you tested it with the update 1.1? ROS might have problems resolving the local IP's and hostnames. Make sure to set all the necessary variables for ROS networking. You can also take a look here: [Network](http://wiki.ros.org/ROS/NetworkSetup)

SimoCerra, 2020-11-25:
Yes, I have updated the app, but it does not work with the hotspot. The ROS variables have been set correctly. 
Thank you for your reply

NRottmann, 2020-11-30:
The connection via a WiFi-hotspot is working as follows:

1. Make sure you have the current version of our application (Version 1.1.1)
2. Set the ROS environment variables on your PC as follows:

````
export ROS_IP=PC_IP                                                     (This is in general 192.168.43.???)
export ROS_MASTER_URI=http://$ROS_IP:11311
export ROS_HOSTNAME=Cellphone_IP                        (Should be 192.168.43.1)
````

3. Set the correct PC IP in the Master Tab of ROS-Mobile (thus probably the 192.168.43.???). Choose the correct device IP in the drop-down menu in the Master Tab (should be the 192.168.43.1). Hit the connect Button and check if everything worked by using the Debug Node.

SimoCerra, 2020-11-30:
1. Following the described steps I was not able to connect to the Master on my PC. 
The following image describes the obtained error:
[image]
On ROS ROS_IP and ROS_HOSTNAME cannot be different, right?

2. May be a problem of ssh authentication?

NRottmann, 2020-11-30:
Please provide for me the output of your terminal py typing 

`ifconfig`

and also your exact ROS export configurations which your inserted.

SimoCerra, 2020-11-30:
[image]

export ROS_IP=192.168.43.99
export ROS_MASTER_URI=http://$ROS_IP:11311            
export ROS_HOSTNAME=192.168.43.1

NRottmann, 2020-11-30:
That seems correct. Did you already check of you can ping your cellphone from your PC as described above:

`ping 192.168.43.1`

SimoCerra, 2020-11-30:
Yes, the PC pings correctly the phone. Indeed, I really don't know what is going wrong. I will try with a standard router instead of Wifi-hotspot and I will let you know.
Thank you for your help

SimoCerra, 2020-11-30:
I have tried the above procedure with a standard router and smartphone-PC are not able to communicate through ROS. May be something related to ssh?
Because the ping works correctly, but when I have typed on my command window: ssh 192.168.1.152 (phone IP) the connection was refused.

SimoCerra, 2020-11-30:
Finally, I have solved. Typing the input in the Master URL I didn't push the 'enter' button when the IP has been set correctly, so the app was not able to store the IP in the related variables.
Thank you so much for your replies!

closed 2020-11-30 by SimoCerra