ROS-Mobile/ROS-Mobile-Android #28: port to ROS2

cocodmdr, 2020-12-01:
Hello,

Thank you so much for this really good application !

I have been using it a bit in ROS 2 but it is a pain to use a ROS bridge to make it work.
Is there a way to port this application to ROS2 without using a bridge ?

Or are there other application out there that are ROS2 compatible?
For now I only need the joystick feature.

Regards,
cocodmdr

NRottmann, 2020-12-02:
Currently, there is no support for ROS2. However, we would love to integrate ROS2 compatibility to our application. Currently, it is planned that we address this issue sometime next year. Nevertheless, we appreciate every contribution regarding this topic that might help us to release a ROS2 compatibility sooner.

Best, Nils

NRottmann, 2021-06-14:
Short update: Due to some issues a ROS2 support is postponed until further notice. As a walk through, consider using [ROS Bridge](https://github.com/ros2/ros1_bridge) for now.

ltbj, 2021-07-01:
@NRottmann can you provide insight to the issues seen?

NRottmann, 2021-07-05:
We are still trouble shooting, but the main problem is that we can not get it to work as suggested in the [ros2-java examples](https://github.com/ros2-java/ros2_android_examples).

kashifkhalid, 2022-01-10:
Following

tsadarsh, 2022-11-25:
> I have been using it a bit in ROS 2 but it is a pain to use a ROS bridge to make it work.

@cocodmdr Can you please give an overview of how you implemented the bridge? Were you running both ROS1 and ROS2 on your host machine?

YasuChiba, 2023-02-06:
Hi all,
I'm currently trying to port ROS-Mobile-Android to ROS2 :) 
Still work in progress, but some widgets are already working!
https://github.com/YasuChiba/ROS2-Mobile-Android

15abhishek, 2023-12-09:
Hi everyone,
I'm new to ROS, can someone help me understand how I can use the app with ROS2 humble?
Thank you in advance!

Wolffy233, 2025-09-26:
Hi everyone,
I'm new to ROS, I need it compatiable for ROS2, but I don`t know how much work it takes, seems a lot,,,