ROS-Mobile/ROS-Mobile-Android #30: Not working, can't connect to ROS master

abougouffa, 2021-01-11:
The app seems to be completely unusable, can't connect either to ROS master or to SSH, the `ip:port` is alright, I even tried to connect from another app (**ROS Control**) using the same phone, and it works!

Here a recording of my screen:

[image]

## Additional Information
### Phone:
POCO X3 NFC (MIUI 12.0.7)

### Android version
10 QKQ1.200512.002

NRottmann, 2021-01-20:
It seems that you did not press the "Enter" Button after inserting the IP-Address. The "Enter" Button has to be pressed such that the IP get locked.

abougouffa, 2021-01-21:
Thank you for your reply @NRottmann. Yep, pressing `ENTER` after inserting the IP address does solve the problem; however, I think that it will be more useful if this step can be omitted; it is counter-intuitive to have to press ENTER on the phone keyboard!, the "Connect" button can easily include the code executed when pressing "Enter."

Making the Connect button does all the needed work to "connect" can increase the usability of the app; I'm sure that many other people have faced the same issue.

Thank you again

NRottmann, 2021-01-22:
We fixed the Issue for the upcoming Version of the App which will be hopefully uploaded to Google Play within the next few days.

closed 2021-01-22 by NRottmann