ROS-Mobile/ROS-Mobile-Android #38: I can't see the map in VIZ tab even ROS shows me the phone is suscribed at topic map

edrsd, 2021-02-14:
Hello, I'm newbie in ROS and I'm doing this tutorial "https://github.com/samialperen/oko_slam/blob/master/doc/hector_slam_tutorial.md"
 about SLAM. I have connected the phone sussesfully to Master and I can move the robot with Joystick but in Viz tab doesn't show the map. I have wroten the rostopic info map command and shows me the phone suscribed to the topic map. what could be going wrong? Thanks in advance.

NRottmann, 2021-02-16:
Probably you did not set the network parameter correctly. Please check out https://github.com/ROS-Mobile/ROS-Mobile-Android/wiki/FAQ on how to set your ROS network parameters correctly. Please then tell us if that solved your problem.

edrsd, 2021-02-17:
thanks a lot you were right. I used this "https://answers.ros.org/question/206876/how-often-do-i-need-to-source-setupbash/" 
to wrote in .bashrc file and add this two lines:
export ROS_IP="MY_PC_IP"
export ROS_MASTER_URI=http://"MY_PC_IP":11311

I have another question: Is there a way to show the robot's position on the map? because for now I can only see how the map is created without seeing a robot reference. Again thank a lot NRottmann.

NRottmann, 2021-02-17:
You are welcome, The feature of showing the robots pose will be added within the next month by my colleague.

closed 2021-03-11 by Torellin

jack162, 2021-07-31:
Hello, could you tell me how to set the network parameter?  i know how to set in pc, but i don't how where is .bashrc file in Android devices. Thanks a lot.

Torellin, 2021-07-31:
There is none. You only have to set the ROS variables on the master Pc and connect to it via the same network. Additionally try to ping your phone/tablet - if this is successful you should be good to go.

jack162, 2021-08-02:
yes,i set the variables as below in master pc

**export ROS_HOSTNAME=192.168.50.77
export ROS_MASTER_URI=http://192.168.50.77:11311
ROS_IP=192.168.50.77**

but in my android device still can't recevice any messenges
(btw, i have try to check the node connection by "rosnode info ..." 
it shows the device node has subscribe to rostopic(/map)   )

NRottmann, 2021-08-02:
Does the joystick not connect or does it not send any data. In the first case, it might be still an issue with the network parameter. In the second case you might have some issues with existing firewalls or network permissions.

jack162, 2021-08-02:
oh,that is second case.
The firewalls or network permissions for pc or android device?