ROS-Mobile/ROS-Mobile-Android #44: Signal plot and battery level

ctampier, 2021-04-09:
Hi and thanks for this great tool.
I just wanted to know if there's any prospects for adding a plot widget (much like rqt_plot).
On the same topic, a robot battery level indicator could be quite useful.

Looking forward to seeing future updates on this project!

Torellin, 2021-04-13:
Great request! We will try to include both of them after the next update (2.0 coming soon)!

ctampier, 2021-04-14:
Much appreciated!
If I may point out another detail, I believe a "deadzone" (maybe a configurable deadzone) in the joystick node would be highly beneficial. Otherwise is very difficult to move the robot in a straight line.

Torellin, 2021-05-13:
https://user-images.githubusercontent.com/37151476/118146510-f6515800-b40e-11eb-9365-9465d5d91d09.mp4

First design of a battery indicator. Would you have any suggestions or things that are missing from the indicator?

MoffKalast, 2021-05-13:
Neat, could possibly have it toggle percentage/voltage text display as an option?

ctampier, 2021-05-14:
Looks good to me :-)

Torellin, 2021-05-25:
> Neat, could possibly have it toggle percentage/voltage text display as an option?

Done! You could potentially try it out by pulling from the Development branch.

closed 2022-11-10 by nicostudt