ROS-Mobile/ROS-Mobile-Android #51: Widget improvements

MoffKalast, 2021-05-13:
Hey, I just gave the thing a go from the play store and I gotta say it's pretty neat!

Some feedback:

- there doesn't seem to be a way to delete added widgets, but I see that's in screenshots here so I assume the store version isn't completely up to date yet?

- instead of having to set the X and Y values of widgets, wouldn't it be better to have a "movement mode" or tab, where widgets would be inactive and you could press and drag them around to make a good layout? would save a lot of time imo

- ssh seems to stop responding to new commands if I run a blocking roslaunch, so there's probably something to think about regarding adding a way to sent a ctrl-c kill interrupt

Looking forward to seeing where you go with this 😄

Torellin, 2021-05-14:
Thank you very much for your Feedback!

1.  The screenshot have to be updated, but you can swipe left while touching the widget or layer to delete them.  Maybe I'll add a small indikator or hint animation for that. 

2.  Have tried to implement that a few times - it is harder than I thought at first so that the widget view and the details are always in sync 😅 I'll keep working on that for sure!

3.  Good suggestion!

Torellin, 2021-05-14:
[image]

MoffKalast, 2021-05-14:
Ah neat, it never occurred to me to swipe the widgets 😅

anfederman, 2021-05-14:
Great minds think alike. I tried almost exactly at the same time.  

1) I got it to move with the joystick. But I didn't see how I could reduce the speed  It wouldn't accept .05  as a value for x, y . Documentation?

2.  I don't have tab completion from the command line so launching the camera from the phone was almost impossible (this might just be my default keyboard on my Pixel phone..

3. Once I launched the camera, I didn't see the display in the viz environment.

All in all a pretty successful first run.  BTW  are you familiar with Robot Commander on the app store and the Ubiquity Robotics github?

stefan-lochbrunner, 2021-05-17:
Hi, I came across the app a few days ago and tried it last week.
- I was also wondering how to delete widgets, tried tapping and holding and I must have only tried swiping in the wrong direction. Tapping and holding the tabs on top already shows a pop up with their name again. Maybe tapping and holding a widget in the details list could give some directions.
- In the end I couldn't quite use the app since I needed to send std_msgs/Float64 messages. It would be great if the message type of the button widgets would accept that. Alternatively, a generic message widget would be great.
- I also had some issues with the app freezing/crashing (on a Pixel 3XL) but can't provide any steps to reproduce it yet. Maybe editing widgets while being connected to a master caused it.
However, I quite like the app and how easy it is to configure. Looking forward to the next version!

MoffKalast, 2021-05-17:
> tapping and holding the tabs on top already shows a pop up with their name again. Maybe tapping and holding a widget in the details list could give some directions.

I'd agree with that, tap and hold to reveal an X button would probably be more intuitive.

> It would be great if the message type of the button widgets would accept that

Well I know that the javascript part of rosbridge is capable of using any message defined as a string. This seems to be a native implementation so I'm not sure if that could also work, but being able to define custom msgs sounds would be valuable in many ways.

NRottmann, 2021-05-18:
> Great minds think alike. I tried almost exactly at the same time.
> 
> 1. I got it to move with the joystick. But I didn't see how I could reduce the speed  It wouldn't accept .05  as a value for x, y . Documentation?
> 
> 1. I don't have tab completion from the command line so launching the camera from the phone was almost impossible (this might just be my default keyboard on my Pixel phone..
> 2. Once I launched the camera, I didn't see the display in the viz environment.
> 
> All in all a pretty successful first run. BTW are you familiar with Robot Commander on the app store and the Ubiquity Robotics github?

1. For the joystick: The Mapping and the Scale factors can be set. For example: Setting for the X-Axis of our joystick the Mapping to Angular and Z, and the Scale to 0.05 and -0.05 means that the output is written to msg.angular.z. and the range from the joystick's x-value ( -1 to +1) is rescaled to 0.05 to -0.05.

2. Tab completion for ROS commands is currently not available. We try to add this.

3. Are the environmental variables set correctly?

Thanks for the Feedback. We are not familiar with Robot Commander or Ubiquity Robotics, but we will have a look into it.

NRottmann, 2021-05-18:
> Hi, I came across the app a few days ago and tried it last week.
>
> * I was also wondering how to delete widgets, tried tapping and holding and I must have only tried swiping in the wrong direction. [...]

Thanks for the Feedback. We will try to add a generic publisher for generating any kind of ROS message.

anfederman, 2021-05-18:
On downloads.ubiquityrobotics.com there is a kinetic vm that would work
with your app.

Also the github has public code. Robot commander supports voice commands. I
think I have a youtube somewhere look for federman robots. I will try to
find the link

Torellin, 2021-05-18:
@anfederman Feel free too implement this feature by forking this project and creating your own widget. 
We will discuss this, but we have to make sure, that all features are free too use and DSGVO conform especially with voice commands. 
Do you have some more information about their software?

Kind regards!