ROS-Mobile/ROS-Mobile-Android #53: BUG?

dineshrboson, 2021-05-25:
[image]
[image]

dineshrboson, 2021-05-25:
[image]
[image]
[image]

dineshrboson, 2021-05-25:
[image]

dineshrboson, 2021-05-25:
I used this app previously with same kind of procedure but now it's little bit confusing can anyone help me out

Torellin, 2021-05-25:
Have you set the environment variables in the bashrc?

Torellin, 2021-05-25:
And maybe try again with Mobile Data disabled. Normally it should work fine, but we have also encountered some issues while being connected to multiple networks simultaneously. For that we have introduced the device ip chooser which you can see in your last image.

dineshrboson, 2021-05-26:
[image]
[image]
[image]
[image]
[image]

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
I hope I tried every possible way.

dineshrboson, 2021-05-26:
[image]

Torellin, 2021-05-26:
Ok this is indeed very strange. Did you have the environment variables before or are they newly added?

dineshrboson, 2021-05-26:
I just added them newly and tried

Torellin, 2021-05-26:
So you have to source the new variables first. As the latest picture indicated, you tried but entered the wrong file. Open a new Terminal should remove this issue.

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
Not working.

Torellin, 2021-05-26:
Ok well.. let me think about this a sec.

Torellin, 2021-05-26:
Are you running roscore on an emulator or virtual environment?  I'm grasping at straws..

dineshrboson, 2021-05-26:
I made my mac into ubuntu and running ros on it.

dineshrboson, 2021-05-26:
no virtual env though

Torellin, 2021-05-26:
Can you ping your phone from your device?

dineshrboson, 2021-05-26:
let me try please

dineshrboson, 2021-05-26:
dinesh@dinesh-MacBookAir:~$ ping 192.168.0.100
PING 192.168.0.100 (192.168.0.100) 56(84) bytes of data.
64 bytes from 192.168.0.100: icmp_seq=1 ttl=64 time=20.5 ms
64 bytes from 192.168.0.100: icmp_seq=2 ttl=64 time=46.2 ms
64 bytes from 192.168.0.100: icmp_seq=3 ttl=64 time=69.8 ms
64 bytes from 192.168.0.100: icmp_seq=4 ttl=64 time=103 ms
64 bytes from 192.168.0.100: icmp_seq=5 ttl=64 time=3.06 ms
64 bytes from 192.168.0.100: icmp_seq=6 ttl=64 time=34.4 ms
64 bytes from 192.168.0.100: icmp_seq=7 ttl=64 time=61.0 ms
64 bytes from 192.168.0.100: icmp_seq=8 ttl=64 time=86.2 ms

dineshrboson, 2021-05-26:
yes I guess

dineshrboson, 2021-05-26:
can you briefly describe steps to connect so that I will try to repeat from starting.

Torellin, 2021-05-26:
In other words, you can confirm all the items in the connection help list in the app?

dineshrboson, 2021-05-26:
valid and active WIFI connections?
YES 
Correct IP and PORT of the ROS MASTER entered?
Don't KNOW  hopefully yes
Is this device in the same network as the Master system?
YES
Changed the environment variables of the Master system?
Dont know hopefully yes
Master is running?
Yes. Roscore

Torellin, 2021-05-26:
Try reinstalling the app. Make sure that you are connected to the same network as your roscore and that only your WiFi on your phone is enabled.
On the Master Pc:
Find your IP with ifconfig, copy it into the ROS env variables in your bashrc. Open a freshly new terminal and try to resolve error messages like the one above by removing the "SLAM"-line from the bashrc and other duplicate lines.
Start the roscore and make sure that the Master IP matches the one entered in the bashrc.
Ping the phone for a quick check.

Enter the correct Master IP on ROS-Mobile.
Now it should hopefully connect.

dineshrboson, 2021-05-26:
can you elaborate ROS_ENV variables ?
is it refers to ROS_HOSTNAME= ip add
or ROS_IP=ip
orROS_Maste r_URI=ip
which of these I need to add?
or all of these?

Torellin, 2021-05-26:
`
myIp=192.168.178.62
export ROS_IP=$myIp
export ROS_HOSTNAME=$myIp
export ROS_MASTER_URI=http://$myIp:11311
`
This is what I have in my bashrc.

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
removed duplicates and error SLAM

Torellin, 2021-05-26:
I don't like to ask, but have you tried other ROS apps for Android and did it work there?

dineshrboson, 2021-05-26:
yeah while I was in my monitor it worked and I am using my personal laptop Its not working

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
You see that this is another app and it is also not working

dineshrboson, 2021-05-26:
previously It worked with another machine

Torellin, 2021-05-26:
Ok, luckily we can confirm that it is not the app ^^

dineshrboson, 2021-05-26:
haha cool :100:

dineshrboson, 2021-05-26:
none of the apps are connecting to my roscore I dont know whats the issue

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
and if I connect though mobile its not connecting

Torellin, 2021-05-26:
What is the content of your "/etc/hosts" file?

dineshrboson, 2021-05-26:
[image]

Torellin, 2021-05-26:
Seems ok I guess. What about running "roswtf" in a second terminal after starting the roscore?

dineshrboson, 2021-05-26:
[image]

dineshrboson, 2021-05-26:
[image]

Torellin, 2021-05-26:
Another thing: Can you confirm the same timezone and clock time of your two devices?

dineshrboson, 2021-05-26:
Yes 12:18 WED MAY26 2021
12:18 WED MAY26 2021
in both devices

Torellin, 2021-05-26:
I'm slowly running out of my expertise. The problem could be that you are running ROS on a laptop and the network setup is somewhat different.

dineshrboson, 2021-05-26:
Yeah That's Fine I will try to resolve the issue.
Thanks for your patience and help

dineshrboson, 2021-05-26:
Hey just to make sure it helps you 
I had firewall in my machine installed longback ufw I looked at rosnetworksetup and disable firewall and uninstalled it and now its working

dineshrboson, 2021-05-26:
Thanks again man.

dineshrboson, 2021-05-26:
please close the issue when you see it.

closed 2021-05-26 by Torellin