ROS-Mobile/ROS-Mobile-Android #54: I want to connect jetson nano and android application via bluetooth.

soyeon219, 2021-05-27:
I want to connect jetson nano and android application via bluetooth. The signal comes from the jetson nano to the smartphone, and I want to run the application according to the signal. Is there any way?

Torellin, 2021-05-27:
Is the signal  in a ROS-compatible format? Then potentially yes.

soyeon219, 2021-05-27:
Yes, we refer to this site for training. 
https://github.com/dusty-nv/jetbot_ros

But I'm not sure how to signal from ros device to smartphone.

Torellin, 2021-05-27:
Please take a look at the ROS Networking Setup on their website. 
http://wiki.ros.org/ROS/NetworkSetup
The App only works if it can connect to a ROS-core and available topics.

NRottmann, 2021-06-14:
Forwarded to ROS network setup.

closed 2021-06-14 by NRottmann