ROS-Mobile/ROS-Mobile-Android #55: Gesture operation caused "pointerIndex out of range"

joye-wang, 2021-05-28:
I used the gesture dector and GridMapView(the bitmap version) which in your project in my source code.
When I triggered a  gesture operation(often scale), use more than two fingers. It's usually crashed. Here is the code caused carsh.

here is the code with problems:

https://github.com/ROS-Mobile/ROS-Mobile-Android/blob/757c04a7bce269bf399efcf29dc3acec114a0d96/app/src/main/java/com/schneewittchen/rosandroid/ui/general/MatrixGestureDetector.java#L61

The error logs :
```
at android.view.MotionEvent.nativeGetAxisValue(Native Method)
        at android.view.MotionEvent.getX(MotionEvent.java:2201)
        at **.widget.map.MatrixGestureDetector.onTouchEvent(MatrixGestureDetector.java:57)
        at **.widget.map.MapView.onTouch(MapView.java:282)
        at android.view.View.dispatchTouchEvent(View.java:11793)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2970)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2657)
        at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:451)
        at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1834)
        at android.app.Activity.dispatchTouchEvent(Activity.java:3375)
        at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:69)
        at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:413)
        at android.view.View.dispatchPointerEvent(View.java:12038)
        at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4859)
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4670)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4206)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4259)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4225)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4352)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4233)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4409)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4206)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4259)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4225)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4233)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4206)
        at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6766)
        at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6740)
        at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6701)
        at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6872)
        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:249)
        at android.view.InputEventReceiver.nativeConsumeBatchedInputEvents(Native Method)
        at android.view.InputEventReceiver.consumeBatchedInputEvents(InputEventReceiver.java:212)
        at android.view.ViewRootImpl.doConsumeBatchedInput(ViewRootImpl.java:6840)
        at android.view.ViewRootImpl$ConsumeBatchedInputRunnable.run(ViewRootImpl.java:6895)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1029)
        at android.view.Choreographer.doCallbacks(Choreographer.java:841)
```

And I found you show the GridMap by using 'opengl' instead of 'bitmap', why? Thanks .

Torellin, 2021-05-28:
Thank you for the feedback! 
Which App/Code Version is this? 

We switched to OpenGL because of the new layer system for the TF View (Viz2D) as its far more easy to work with the camera system of OpenGl together with frame transformations.

joye-wang, 2021-05-31:
The `build.gradle` in my project is:
```
android {
    compileSdkVersion 29
    buildToolsVersion "30.0.2"

    defaultConfig {
        applicationId "joye.wang.rosclientapp"
        minSdkVersion 25
        targetSdkVersion 29
        versionCode 1
        versionName "1.0"

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }

    buildFeatures {
        dataBinding true
    }
}
```

The device version is `android 8.1.0`,  This bug often appears in multi finger operation(more than two finger).

Using opengl is a wisdom choice, but I think the communicate method(rpc xml) of `ros-android` isn't a wisdom choice. I'm using  rosbridge(websocket json) instead of xml. And the opengl version of `ros-android-app` is opengl 1.0.

Do you have any plan of update the ros-package, such as websocket, opengl version and `point cloud` support?

Thanks for your reply~

Torellin, 2021-05-31:
We might switch to websockets due to my unsuccessful attempts with ROS2 for android implementation. Until now I really have no idea how to build the project including rcljava without errors. It's the first time I've worked with OpenGL, but as I understand it, the minimum version of Android 5 we support should also work with OpenGL ES 3.1. That would definitely be worth a try.
We will add PointClouds once we have implemented the 3D transformation view. Currently the focus is on the development of the most common messages for 2D TF view.
So lot of work to come ;)

joye-wang, 2021-05-31:
Sorry, I forget one thing: The bug is appeared when I used the MapView(draw by canvas), not OpenGL MapView.

NRottmann, 2021-06-14:
Solved within newer App version.

closed 2021-06-14 by NRottmann