ROS-Mobile/ROS-Mobile-Android #57: Support for Diagnostics

blubbi321, 2021-06-08:
Thanks for all your great work so far!

It would be great if you could also add support for diagnostic messages to the app. (http://wiki.ros.org/diagnostics)

Torellin, 2021-06-09:
Thank you for your kind feedback! 
I will have a look at it. Until then, you could use the debug widget and display the diagnostic msgs directly as textform. Better than nothing I guess.

Torellin, 2021-06-09:
Do you have any ideas or wishes how to display the diagnostics?

blubbi321, 2021-06-09:
I think the rqt robot monitor (https://github.com/ros-visualization/rqt_robot_monitor) does a good job in general with its tree display. 
For a space constrained mobile display I would probably even leave out the two separate displays for warnings and error. 
If the tree is too much for a first step that list with warnings and errors would help already ;-)

Torellin, 2021-06-10:
I will give it a try and maybe in a few days present the first draft for the visuals here.