ROS-Mobile/ROS-Mobile-Android #58: GridMapView stride issue

zjn0505, 2021-06-11:
Hey, thanks for this good app, it helps me understand ROS messages as a beginner.

I'm working on drawing OccupancyGrid on Android, and taking your GridMapView as reference.

A small issue I saw in the GridMapView is that `tile.setStride()`on current master branch is not correct.

https://github.com/ROS-Mobile/ROS-Mobile-Android/blob/757c04a7bce269bf399efcf29dc3acec114a0d96/app/src/main/java/com/schneewittchen/rosandroid/widgets/gridmap/GridMapView.java#L106

The stride should be the width of data in that tile, not the width of the entire grid.

I changed it into something like
```
val isLastColumn =  x == numTilesWide - 1
val stride = if (isLastColumn) {
    width.mod(TextureBitmap.STRIDE)
} else {
    TextureBitmap.STRIDE
}
tiles[tileIndex].setStride(stride)
```       

Also in this class, all the `TextureBitmap.STRIDE` should be `TextureBitmap.HEIGHT` when used with `y`.

Torellin, 2021-06-13:
Thank you for your feedback. 
I will take a look on this :)

zjn0505, 2021-06-14:
Another small issue in `CameraControl` class.
If I init it with `cameraControl.init(translate = true, rotate = false, scale = true)`, it won't do translate and scale either.
The `onTouchEvent` currently is asking for all 3 gestureDetectors.

Torellin, 2021-06-14:
The camera control bug is fixed and the stride thingy seems not to change anything as the variable is static anyway. I will close this issue, but if you find any new infos, please feel free to reopen it.

closed 2021-06-14 by Torellin

Torellin, 2021-06-14:
I think it really depends, how you expect the map to look like. As long as nothing breaks, it should be fine.

zjn0505, 2021-06-15:
Thanks for the update.

The GridMap will look distorted if `TextureBitmap.STRIDE` is used for `y`, and its value got changed differently from `TextureBitmap.HEIGHT`, I did this this change locally and found the distort issue. If `TextureBitmap.STRIDE` is same as `TextureBitmap.HEIGHT`, then all good.

`tiles.get(tileIndex).setStride(width)` with width of entire map will also render a distorted map on the last column when the map's width is larger than `TextureBitmap.STRIDE`.