ROS-Mobile/ROS-Mobile-Android #62: Crash on connecting to server

blubbi321, 2021-06-30:
I got an auto update of the app (2.1) to my phone (Samsung S8) yesterday.

Connecting to a ros master now crashes the app immediately. Tried to reinstall but the issue persists.

Let me know if I can assist with log etc in case you cannot reproduce the issue.

NRottmann, 2021-06-30:
Hi, can you post here your log data. I can not reproduce your error. In our tests, ROS-Mobile connects always with the master.

blubbi321, 2021-07-01:
Sure Im happy to provide it if you tell me what to obtain (and how) - Apologies, not an Android developer ;)

NRottmann, 2021-07-02:
Please use Android Studio. First connect your smartphone to your pc via USB:

[image]

Second, run current version of the app and check the log. Make sure you picked the correct program running:

[image]

Third, send use the complete error message (will be marked red).

cahidenes, 2021-08-12:
I have the same problem, when I press the connect button, the log output is as follows:

    I/RosRepository: Connect to Master
        I/System.out: [socket]:check permission begin!
        W/System: ClassLoader referenced unknown path: system/framework/mediatek-cta.jar
        I/System.out: [socket] e:java.lang.ClassNotFoundException: com.mediatek.cta.CtaUtils
        I/RosRepository: Update Master
        I/RosRepository: Register Node: /tf_static
        I/RosRepository: Register Node: /tf
        W/chen.rosandroi: Accessing hidden field Lsun/misc/Unsafe;->theUnsafe:Lsun/misc/Unsafe; (greylist, reflection, allowed)
        W/chen.rosandroi: Accessing hidden method Lsun/misc/Unsafe;->objectFieldOffset(Ljava/lang/reflect/Field;)J (greylist,core-platform-api, linking, allowed)
            Accessing hidden method Lsun/misc/Unsafe;->compareAndSwapObject(Ljava/lang/Object;JLjava/lang/Object;Ljava/lang/Object;)Z (greylist, linking, allowed)
            Accessing hidden method Lsun/misc/Unsafe;->compareAndSwapInt(Ljava/lang/Object;JII)Z (greylist, linking, allowed)
            Accessing hidden method Lsun/misc/Unsafe;->getUnsafe()Lsun/misc/Unsafe; (greylist,core-platform-api, linking, allowed)
        I/Registrar: MasterXmlRpcEndpoint URI: http://192.168.1.113:11311/
        I/Registrar: MasterXmlRpcEndpoint URI: http://192.168.1.113:11311/
        E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-13
            Process: com.schneewittchen.rosandroid, PID: 13194
            org.ros.exception.RosMessageRuntimeException: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid--9vEBZctUfmhCiHzDRSU4A==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid--9vEBZctUfmhCiHzDRSU4A==/lib/arm64, /system/lib64, /product/lib64]]
                at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:58)
                at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44)
                at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
                at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
                at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
                at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
                at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
                at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
                at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
                at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
                at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
                at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
                at org.ros.internal.message.topic.TopicDescriptionFactory.newFromType(TopicDescriptionFactory.java:36)
                at org.ros.internal.node.DefaultNode.newPublisher(DefaultNode.java:284)
                at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41)
                at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187)
                at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85)
                at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173)
                at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
                at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
                at java.lang.Thread.run(Thread.java:919)
             Caused by: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid--9vEBZctUfmhCiHzDRSU4A==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid--9vEBZctUfmhCiHzDRSU4A==/lib/arm64, /system/lib64, /product/lib64]]
                at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:196)
                at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
                at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
                at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:54)
                at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44) 
                at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
                at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
                at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107) 
                at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86) 
                at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49) 
                at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
                at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
[... cut ...]
                at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187) 
                at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85) 
                at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173) 
                at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
                at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
                at java.lang.Thread.run(Thread.java:919)

atfkerem, 2022-10-18:
Same problem here. After succesfully establishing SSH connection, we I try to establish master connection app crashes.
`2022-10-18 09:37:58.510 21091-22456/com.schneewittchen.rosandroid E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-2
    Process: com.schneewittchen.rosandroid, PID: 21091
    org.ros.exception.RosMessageRuntimeException: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/lib/arm, /system/lib, /system/vendor/lib]]
        at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:58)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44)
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
        at org.ros.internal.message.topic.TopicDescriptionFactory.newFromType(TopicDescriptionFactory.java:36)
        at org.ros.internal.node.DefaultNode.newPublisher(DefaultNode.java:284)
        at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41)
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187)
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85)
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/lib/arm, /system/lib, /system/vendor/lib]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:134)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:54)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44) 
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107) 
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86) 
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49) 
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107) 
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86) 
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49) 
        at org.ros.internal.message.topic.TopicDescriptionFactory.newFromType(TopicDescriptionFactory.java:36) 
        at org.ros.internal.node.DefaultNode.newPublisher(DefaultNode.java:284) 
        at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41) 
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187) 
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85) 
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) 
2022-10-18 09:37:58.510 21091-22468/com.schneewittchen.rosandroid E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-12
    Process: com.schneewittchen.rosandroid, PID: 21091
    org.ros.exception.RosMessageRuntimeException: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid-5WHMOHwKfEOeZHozejHmwA==/lib/arm, /system/ [... cut ...]
        at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41) 
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187) 
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85) 
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) 
`

p-Psquare, 2025-03-26:
I encountered similar connection issues when running the app natively. However, when I used it [here](https://nativebridge.io/app/GeVa)  to run the app directly in my browser, the connection was stable, and the app functioned smoothly. It might be worth exploring this as an alternative testing environment.