ROS-Mobile/ROS-Mobile-Android #63: GirdMap isn't Display For  origin: [-5.250000, -5.050000, 0.000000]  of file .yaml

caiyahui, 2021-07-09:
In navigation mode，For the topic of /map ,GirdMap isnot Display;

Modify the contents of  file .yaml, 
Change  origin: [-5.250000, -5.050000, 0.000000]  of file .yaml to origin: [-0.000000, -0.000000, 0.000000]  of file .yaml ,the topic of /map ,GirdMap is Display OK;

Excuse me, why can't  origin: [-5.250000, -5.050000, 0.000000]  of file .yaml  ,the topic of /map ,GirdMap isn't Display

Torellin, 2021-07-09:
This seems to be a general ROS question/comment and not particular about the app, or do I get this wrong?

caiyahui, 2021-07-10:
the problem of origin: [-5.250000, -5.050000, 0.000000] of file .yaml  is that the Android Apps Map NAV does not exist，Will it be the problem of the app gird map  analysis

Torellin, 2021-07-11:
Can you send the content if this yaml-file?

caiyahui, 2021-07-12:
Hi All:

The attachment for the map.pgm and map.yaml

caiyahui, 2021-07-12:
Hi

Map

caiyahui, 2021-07-12:
Use the App of ROS-Mobile-Android and  Map NAV to display different effects for  the File of map.pgm and map.yaml;
the App of ROS-Mobile-Android  display :
[image]
[image]
[image]
the App of  Map NAV  display :
[image]

the File of map.pgm and map.yaml
[map.zip](https://github.com/ROS-Mobile/ROS-Mobile-Android/files/6799318/map.zip)

Torellin, 2021-07-12:
The laserscan seems ok. Is the map subscribed? Can you share the Widgets list as well as the Layer list of the TF Viz in the Details Tab?

Torellin, 2021-07-13:
Alright, can you confirm that on the \map-topic a nav_msgs/OccupancyGrid is published? The App doesn't know about your yaml files. It only subscribes to the map-topic and also get all necessary infos from the message. If there is something missing, the app cannot display the map correctly.

caiyahui, 2021-07-13:
[image]
[image]

NRottmann, 2021-07-15:
We have an idea what the problem might be. We will check it within the next days and come back to you.

NRottmann, 2021-07-20:
Please try out the current Development Branch. We hope that this fixed your problem. For building the app from source, follow the instructions here: https://github.com/ROS-Mobile/ROS-Mobile-Android#built-from-source

caiyahui, 2021-07-21:
The result is the same ，Try out the current Development Branch.   

Sync with apply from: "https://github.com/rosjava/rosjava_bootstrap/raw/kinetic/buildscript.gradle"

Where is the change between the current Development Branch and before.

NRottmann, 2021-07-21:
For messages like \map, most often we only receive a message at the start of the node subscription. The problem is that we are not storing any last message data, thus at the first initialization of the view the map data is received and the map is displayed. By changing now the tab and going back, the view is destroyed and restored. But since we already received a message, we can not restore the data. To solve this issue, we now store the last message for a subscriber such that we can rebuild the view.

I thought that this could have been your problem, but it seems that there is another. My best guess now is that there is an issue with the TFs.

caiyahui, 2021-07-23:
In one case, /map is displayed，The car at two points in the actual navigation process。
But I stopped navigating, or didn't start navigating, changed messages  \map，After adjusting back to  \map it also does not show。
In the case of \map display, WIGETS multiple layers will also cause the map to not be displayed

caiyahui, 2021-07-23:
The car at two points in the actual navigation process。
messages \map display error, messages \move_bass display ok
[image]
[image]
[image]

NRottmann, 2021-07-23:
Do you have a GitHub Repo from which we can download your simulation to reproduce the error?

caiyahui, 2021-07-24:
Just use the latest version，What's the difference between  messages \map and messages \move_bass

NRottmann, 2021-07-26:
In order to be able to help you solve your problem, we need to recreate the error. Therefore, we need you simulation environment and a description how to reproduce the error. Then we can debug it and probably solve it.

\map and \move_base are not message, they are topics which provide some specific messages. \map provides thereby most often a nav_msgs/OccupancyGrid message.

caiyahui, 2021-07-26:
Hi NRottmann:
Thank you for following up, Part of the content is not displayed;
Use your latest ROS-Mobile-Android app-release.apk testing;

GUI for Details and Widgets :
[image]

file of .pgm and .yaml
[map.zip](https://github.com/ROS-Mobile/ROS-Mobile-Android/files/6877393/map.zip)

mapping  mode:
[image]

nav. mode(Circular navigation between two points):
display ok:
[image]

display error:
[image]

caiyahui, 2021-09-21:
> In navigation mode，For the topic of /map ,GirdMap isnot Display;
>
> Modify the contents of file .yaml,
> Change origin: [-5.250000, -5.050000, 0.000000] of file .yaml to origin: [-0.000000, -0.000000, [...]

closed 2021-09-21 by caiyahui