ROS-Mobile/ROS-Mobile-Android #64: Related questions about the button function

ju-mingyue, 2021-07-26:
@Cycov 
Hello, for this software, the function is relatively complete at present, but the fly in the ointment is that the bool-type button of the button is not continuous to the message, so I cannot judge the message of the button during the use, so can I Add a function that can continuously publish information

NRottmann, 2021-07-26:
We already added this kind of function but in the current version it is enabled. Do you require a message received by the first subscription to the topic or do you require a constant message flow, lets say with a constant publishing rate, e.g. 20 Hz.

ju-mingyue, 2021-07-26:
@NRottmann 
Yes, I may be a constant message flow that you need to say. At present, I publish topics from the mobile terminal, and the content of the topic is True/False when viewed on the computer side, but I cannot do it when I make an if judgment. How do I set this up?

NRottmann, 2021-07-26:
Normally, on the computer side you would store the current boolean value in your node and only change it if you received a new message. With that, you can simply access the current status at any time.

ju-mingyue, 2021-07-26:
Dear Mr.NRottmann , I am very sorry that I did not fully understand what you mean. Can you elaborate on how I made this judgment? Because this is my first contact with this field.

NRottmann, 2021-07-26:
Normally, you have a ROS-Node which runs on your computer. This ROS-Node initializes a Subscriber whoch subcribes to the Button topic. A possible option is now, for example, to create a class where your subscriber as well as a private Boolean is inherited such that you simply update the Boolean variable within your subscriber's callback function.

ju-mingyue, 2021-07-27:
@NRottmann 
Thank you very much for your prompt reply, I have some understanding of your method. At the same time, I am also very sorry to trouble you. The following is the program I wrote according to your intentions. I am not sure whether this is correct or whether it can express your meaning correctly.
                    `
                            import rospy
                            from std_msgs.msg import Bool
                            from geometry_msgs.msg import Pose, PoseWithCovarianceStamped, Point, Quaternion, Twist
                    
                    class Buttons():
                     def __init__(self,Bool):
                       rospy.init_node('button')
                    rospy.Subscriber("/button0",Bool,self.Callback1)
                    rospy.Subscriber("/button1",Bool,self.Callback2)
                    rospy.Subscriber("/button2",Bool,self.Callback3)
                    self.data =True
                    
                    class Move(Buttons):
                     def __init__(self,Bool,Pose):
                       super(Move, self).__init__(Bool)
                    locations = dict() 

				locations['1'] = Pose(Point(0.865, -0.245, 0.000),  Quaternion(0.000, 0.000, -0.561, 0.828))  
				locations['2'] = Pose(Point(1.784, -5.144, 0.000),  Quaternion(0.000, 0.000, -0.645, 0.764))  
				locations['3'] = Pose(Point(2.121, -7.942, 0.000), Quaternion(0.000, 0.000, 0.729, 0.685))  
		def Callback1(self,Pose):
    			self.msg = Pose()
				self.msg.Pose = Pose.locations['1']
				global msg_pub
				self.msg_pub.publish(self.msg)
				self.send(event)
				self.data = False
`
Regarding class inheritance, I don't know how to use it, because python is also a coding language that I just learned, so I am not very familiar with it. At present, this button problem has troubled me for two weeks. If you can guide me to solve this problem in your free time, I will be very grateful to you.

NRottmann, 2021-07-27:
````
#!/usr/bin/env python
import rospy
import sys
from std_msgs.msg import Bool

class simple_class:

  def __init__(self):
    self.btn_state = false
    self.sub = rospy.Subscriber("btn_press", Bool, self.callback)

  def callback(self,data):
    self.btn_state = data.data

def main(args):
  obc = simple_class()
  rospy.init_node('simple_class', anonymous=True)
  try:
    rospy.spin()
  except KeyboardInterrupt:
    print("Shutting down")

if __name__ == '__main__':
    main(sys.argv)
````

That should do the work. It suscribes to the topic btn_press and stores the data within the class variable btn_state.

ju-mingyue, 2021-07-27:
@NRottmann Thank you very much, my problem has been solved very well.

closed 2021-07-27 by ju-mingyue