ROS-Mobile/ROS-Mobile-Android #66: Joystick not working

GodMiu-6789, 2021-08-04:
Dear team,
I just installed the app on my android emulator (android 5.1). I am using raspberry pi running on Ubuntu 18.04 (supported by ros melodic). I am trying to use joystick to teleop my robot .However, after successfully connecting the app with my pi (I can do rostopic list and i see /cmd_vel ), the joystick seems not to publish any Twist msg (I echo /cmd_vel). Can anyone help me please ? I really appreciate it !
Regards,
GodMiu

NRottmann, 2021-08-04:
Is the joystick registered to the /cmd_vel topic? Thus, if you typing
````
rostopic info /cmd_vel
````
does the joystick node appear as publisher? If so and when no messages are poblic, make first sure that all ROS network parameters are set probably.

GodMiu-6789, 2021-08-04:
Dear NRottmann,
Thank you for your response! Yes I did rostopic info /cmd_vel and there is a node called /cmd_vel shown as a publisher. I guess that is what you meant by joystick node. Can you explain more on what should be the network parameters to be set. I 
Regards,
GodMiu

GodMiu-6789, 2021-08-04:
I also want to add that I did set the Master Url and the right port.

GodMiu-6789, 2021-08-04:
[image]
To make it more clear, this the what I see in terminal

GodMiu-6789, 2021-08-04:
[image]
This is what I see in terminal.

NRottmann, 2021-08-11:
The network parameter settings are explained here
https://github.com/ROS-Mobile/ROS-Mobile-Android/wiki/FAQ

amjack0, 2022-04-13:
May be try, deleting the Joystick widget and add again. This worked for me !

GodMiu-6789, 2022-04-13:
Thanks man, I have completed my project for 8 months with a high mark. Btw, thank you for ur recommendation!

closed 2022-11-09 by nicostudt