ROS-Mobile/ROS-Mobile-Android #67: Issue with compressed image subscription

liver121888, 2021-08-08:
I am testing a webcam for my robot. I am using a virtual machine with ubuntu 18.04 to ssh the actual robot. The robot is jetson nano equipped with ubuntu 18.04, too. Now I was trying to hook up the app to the machine. The /cmd_vel works fine, but the camera is not working. I check for some issues on the board and I make sure that ufw is down.
First, my cam is working(view from virtual ubuntu):
[image]
And here is my config(clbrobot is the real robot, while ubuntu is the virtual machine, g3226 is my phone):
[image]
the machine can ping each other(I am not familiar with android so I don't know how to ping in android environment)
[image]
rosnode list and rostopic list
[image]
rosgraph and roswtf
[image]
[image]
so when I get to the info I get this:
[image]
The node can contact each other however the topic is not transmitted inbound to the node. I am confused.
I am having a hard time looking for the answer, they always link to the network and multiple machine tutorial page.
It is my first time writing stuff like this so maybe the question sounds silly but please help me because I have been stuck here for like two weeks. 😢

liver121888, 2021-08-08:
And here is the config on the phone:
[image]
the phone can ssh clbrobot
[image]
camera setting
[image]

Torellin, 2021-08-11:
I can only guess what the problem could be, but probably it is your exported ROS_MASTER_URI with your local address "clbrobot". Please try to follow the convention with 
```
ROS_IP=<masterIP>
ROS_HOSTENAME=<masterIP>
ROS_MASTER_URI=http://<masterIP>:11311
```
on ther Master Pc.
Also make sure you can confirm all other bullet points which pop up inside the App on the Master page.

liver121888, 2021-08-17:
The problem is solved, and everything works fine. Thank you for helping me out!

closed 2021-08-17 by liver121888

PlainJi, 2021-11-21:
same problem and solved. Tks!
As ROS mobile can't resolve the self-defined hostname, we should use IP directly for ROS_HOSTNAME.
Doesn't matter for ROS_IP and ROS_MASTER_URI.