ROS-Mobile/ROS-Mobile-Android #71: Widget idea

MoffKalast, 2021-09-09:
I've been using the app on a project for some time now, and the thing I've noticed seems to be an issue is sending custom data from the app that isn't a Bool. Say if you want to switch modes, or adjust a number value higher and such.

An interesting way to do that I think would be to have a knob, where you could specify the ticks. A basic one would be say 1, 2, 3, 4, 5, etc. but having something like HOME, NAV, STOP, TELEOP or something of the sort. I'm not sure if it would be viable to have the user hold and drag to rotate it, but even just tapping it and having it increment and sending the current setting as a topic publish would be great. That way one could modify more complex settings. Doing it all as a std_msgs/String would probably be good enough.

[image]

Of course it would also be neat if we could have buttons send a custom message or call a service onpress only (and not have it send anything on unpress like it does now), with an option to latch.

Torellin, 2021-09-20:
Same here - thanks again for the suggestions! 
Services would be a great deal for the app in general. 
I really like the knob idea and will try to implement it.

MoffKalast, 2021-09-20:
Thanks for all the great work, it's slowly turning into a really indispensable tool to have around. 😄