ROS-Mobile/ROS-Mobile-Android #73: The application closes when I press the connect button.

VacumCleaner, 2021-09-17:
I want to realize the simulation environment on the computer with control the mobile application. But when I click on the connection button, the text connected is displayed and the application closes. How can i solve this problem.
I will be glad if you help me.

VacumCleaner, 2021-09-17:
This is error 

`I/RosRepository: Connect to Master
I/RosRepository: Update Master
I/RosRepository: Update Master
V/StudioTransport: Handling agent command 1200 for pid: 11563.
I/RosRepository: Register Node: /tf_static
    Register Node: cmd_vel
I/RosRepository: Register Node: /tf
W/chen.rosandroi: Accessing hidden method Lsun/misc/Unsafe;->getUnsafe()Lsun/misc/Unsafe; (greylist,core-platform-api, linking, allowed)
I/Registrar: MasterXmlRpcEndpoint URI: http://192.168.1.55:11311/
I/Registrar: MasterXmlRpcEndpoint URI: http://192.168.1.55:11311/
I/Registrar: MasterXmlRpcEndpoint URI: http://192.168.1.55:11311/
E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-19
    Process: com.schneewittchen.rosandroid, PID: 11563
    org.ros.exception.RosMessageRuntimeException: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid-IJjIPh9OT-kAT3wxWr6DFw==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid-IJjIPh9OT-kAT3wxWr6DFw==/lib/arm64, /system/lib64, /system/product/lib64]]
        at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:58)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44)
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61)
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172)
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107)
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49)
        at org.ros.internal.message.topic.TopicDescriptionFactory.newFromType(TopicDescriptionFactory.java:36)
        at org.ros.internal.node.DefaultNode.newPublisher(DefaultNode.java:284)
        at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41)
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187)
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85)
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "std_msgs.uint32" on path: DexPathList[[zip file "/data/app/com.schneewittchen.rosandroid-IJjIPh9OT-kAT3wxWr6DFw==/base.apk"],nativeLibraryDirectories=[/data/app/com.schneewittchen.rosandroid-IJjIPh9OT-kAT3wxWr6DFw==/lib/arm64, /system/lib64, /system/product/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:230)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at org.ros.internal.message.definition.MessageDefinitionReflectionProvider.get(MessageDefinitionReflectionProvider.java:54)
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:44) 
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107) 
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86) 
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49) 
        at org.ros.internal.message.Md5Generator$1.variableValue(Md5Generator.java:61) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parseField(MessageDefinitionParser.java:172) 
        at org.ros.internal.message.definition.MessageDefinitionParser.parse(MessageDefinitionParser.java:107) 
        at org.ros.internal.message.Md5Generator.generateText(Md5Generator.java:86) 
        at org.ros.internal.message.Md5Generator.generate(Md5Generator.java:49) 
        at org.ros.internal.message.topic.TopicDescriptionFactory.newFromType(TopicDescriptionFactory.java:36) 
        at org.ros.internal.node.DefaultNode.newPublisher(DefaultNode.java:284) 
        at org.ros.internal.node.RosoutLogger.<init>(RosoutLogger.java:41) 
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187) 
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85) 
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:919) 
E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-18
[... cut ...]
        at org.ros.internal.node.DefaultNode.start(DefaultNode.java:187) 
        at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85) 
        at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:919) 
I/Process: Sending signal. PID: 11563 SIG: 9
`

Torellin, 2021-09-20:
What widgets do you have added to the Viz tab? And have you cloned the project or installed via Play Store?
Best regards

VacumCleaner, 2021-09-22:
It closes even though I do not install anything on the Viz screen. The application I installed from the Play store also closes due to the same error.
Best regards

closed 2021-09-26 by VacumCleaner