ROS-Mobile/ROS-Mobile-Android #74: draw paint with antialias on canvas

WuYeaR, 2021-09-27:


Torellin, 2021-09-27:
Okay, can you explain where you would enable the anti aliasing?  I would appreciate a little more details to look into it.

WuYeaR, 2021-09-27:
> Okay, can you explain where you would enable the anti aliasing? I would appreciate a little more details to look into it.

for example, i add antialias in textPaint in LabelView.java
[image]
then widgets with antialiasing are look like:
[image]
it will look better but I'm not sure if it will increase the load on the cpu

Torellin, 2021-09-27:
For sure it will decrease performance but we could let the user decide via an option to enable better graphics for example with anti-aliasing.
Thank you!