ROS-Mobile/ROS-Mobile-Android #81: Compressed image not showing

P4STA, 2021-12-17:
Hello, I'm publishing a compressed image from a raspicam_node on my robot. 
I can see the image through rqt image view widget, so I know it is being published. 
However, the app is showing no image. The topics are set up properly. Connection works as well since I can publish cmd_vel with the joystick. 

Any idea what could be the cause?

nicostudt, 2021-12-20:
Hey,

There are multiple possible reasons for this issue. The obvious one would be a simple connection error in some way. To check that, you could try to subscribe to another topic that is not an image like a Twist message or a simple String message with the Logger/Debug widget and then check if that messages are received.

nicostudt, 2021-12-20:
In most cases that we know where one can send but not receive messages there is an issue with the system ROS variables which has to be set like the MASTER_URI.

P4STA, 2021-12-20:
Yes, it was the ROS_MASTER_URI. I've used an already made SD card image with preinstalled ROS. And even though I set up ROS_MASTER_URI and ROS_IP by hand into the .bashrc, the roscore still refused to start with those parameters. I had to disable some startup scripts before it started working. So nothing related to ROS-mobile after all.

closed 2021-12-20 by P4STA