ROS-Mobile/ROS-Mobile-Android #83: More versatile buttons

P4STA, 2022-01-18:
Hello, 
currently the button widget publishes a Bool message. If I want to use multiple buttons I need to create a separate topic for each button. With increasing number of buttons the amount of topics creates unnecessary clutter.
I would be great if the buttons could send an integer or string message type, so that I could put them all on one topic and subscribe to it. 

Example: I have a mobile robot that can travel to 9 different waypoints on a map. I have a UI with 9 buttons corresponding to those waypoints. I want to press a button and send a corresponding waypoint number to my navigation stack. 

Thanks

PS: great app!

nicostudt, 2022-01-18:
Hey,
Great idea! One could think of the general potential of this, that a widget could also set/init one part of a message instead of one whole message.

nicostudt, 2022-01-18:
The problem of your approach is probably the needed use of a custom message, which is right now, not supported.

P4STA, 2022-01-18:
I think std_msgs/Int16.msg, std_msgs/Int32.msg, or std_msgs/String.msg would be suitable enough for this. So no need for custom messages, only the data inside would need to be customised for each button.