ROS-Mobile/ROS-Mobile-Android #88: Unable to drive the mobile base forward and reverse due to thin GUI interface

amjack0, 2022-04-13:
Hi,

I am using the ROS Mobile app for publishing the `/cmd_vel` to drive the mobile base. I am able to connect the app with the master and able to drive the robot. 
My issue is `I am unable to drive the robot in a pure forward and backward movement`. Because I can not keep the round knob in a straight line with the **VERY THIN VERTICAL AXIS** in a GUI.

Am I missing something here ? 

ROS Noetic + 20.04 LTS
Android device: Samsung Galaxy Tab S T800 [https://www.amazon.de/Samsung-Tablet-PC-interner-Speicher-Bluetooth-grau/dp/B00R48MCFC](url)

nicostudt, 2022-04-15:
Thanks for pointing this out! 
As you said, it is nearly impossible to keep the knob at a zero degree angle to the vertical axis. 
I had the idea to introduce a new joystick mode like a "sticky axis" which would publish only one of the axis values.

In the meantime you could filter the axis you dont want on the ROS side.

amjack0, 2022-04-19:
Hi @nicostudt, 
Thanks for your support.

`In the meantime you could filter the axis you dont want on the ROS side.`
Can you explain, What do you mean by this ?

nicostudt, 2022-04-19:
For example write an additional node that republishes /cmd_vel on a new topic but only with the values you want.

nicostudt, 2022-04-19:
Give me a minute. I'll try to write you the necessary code for the node.

nicostudt, 2022-04-19:
Of cause you can change the input/output topics, the axis you use and the "threshold". The threshold simulates the area in the picture you mentioned above.
I assumed both of your axis have values between -1 and 1, but feel free to remove the threshold variable and put in your numbers.

nicostudt, 2022-04-19:
> both of your axis have values between -1 and 1
... and are centered around 0

nicostudt, 2022-04-19:
Gern geschehen :)

amjack0, 2022-04-19:
I resolved the issue with,

take the turn only when, `if (abs(msg.angular.z) > abs(msg.linear.x))`: is true. Else I drive forward and backward. 
With this I can do PURE forward and backward motion. 

You can close the issue, if you like.
Thank you ;)

closed 2022-04-19 by nicostudt