ROS-Mobile/ROS-Mobile-Android #90: Logger component not updating

Peppe21, 2022-05-18:
Hello,
I'm unable to get the *Logger* component properly working. My expectation is that once added and content published into the /log topic I should get published content displayed in the component.

I ensured node connectivity by running `rostopic info /log`. it properly indicate that ros mobile is subscribing to the topic : 
*Type: std_msgs/String*

*Publishers: None*

*Subscribers:*
\* */log _(http://192.168.1.150:39469/)_*

However, when publishing content like this : `rostopic pub /log std_msgs/String "Test"`
I would expect to see "Test" displayed on the app logger component, but nothing happen.
Thanks for your support

nicostudt, 2022-05-19:
Hey,
this is what I also would expect.
Have you tested if other widgets are working in general?
If this is not the case, it might solve your problem to confirm the checklist, which opens when you try connect the app to your PC / robot without a running Master on it.
There are various problems listed like e.g. "Is your master running?", "Do you have set the environment variables?", etc.

Please feel free to ask, if you have problems with the checks!

Peppe21, 2022-05-21:
Thank you for reverting!

The joystick and buttons are already implemented and working as expected with other nodes. But logger not  and it  is a subscriber only component.

Master environment variable were not touched.
Master in the same subnet.

As rostopic is showing the subscription up, I was suspecting a handler issue in the component itself. 

Logger is a nice to have to set status over button pressed. Would be nice to have it working !

P.

maxiew123, 2022-05-23:
> Thank you for reverting!
>
> The joystick and buttons are already implemented and working as expected with other nodes. But logger not and it is a subscriber only component.
>
> Master environment [...]

Hello, I have the same issue. I can publish messages to the master from my mobile app but I can not subscribe to them. I also changed my environments variables (from https://github.com/ROS-Mobile/ROS-Mobile-Android/wiki/FAQ). Did you resolve it Peppe?

Peppe21, 2022-05-23:
> Hello, I have the same issue. I can publish messages to the master from my mobile app but I can not subscribe to them. I also changed my environments variables (from [...]

Nope.  Any yes, I ensured environment variable as well.

Diiliev, 2022-05-24:
Hello @Peppe21,
My guess is that the terminal window from which you are publishing does not have the correctly set environment variables. You can check using the `echo` command.

Try the following steps:
1. Connect your PC to a stable Wi-Fi network or even a Wi-Fi hotspot from your phone.
2. Open a new terminal window.
3. Find your ip address using:
`$ ifconfig`
Lets say for the purposes of this example that your ip address is: 192.168.43.48
4. Set your environment variables for the current terminal window:
`$ export ROS_IP=192.168.43.48`
`$ export ROS_MASTER_URI=http://$ROS_IP:11311`
`$ export ROS_HOSTNAME=$ROS_IP`
5. Start the ROS Master
`$ roscore`
6. Connect your phone to the same Wi-Fi network. This is not necessary if it is hosting a Wi-Fi hotspot.
7. Select ROS Mobile Configuration
8. Open the MASTER tab
9. Enter 192.168.43.48 into the Master URL field
10. Master port should be 11311
11. Click on CONNECT
12. Open the VIZ tab.
13. Back on your PC, open a new terminal window
14. Set the environment variables for this window as well by repeating step 4.
`$ export ROS_IP=192.168.43.48`
`$ export ROS_MASTER_URI=http://$ROS_IP:11311`
`$ export ROS_HOSTNAME=$ROS_IP` 
15. Publish your message to /log
`$ rostopic pub --once /log std_msgs/String "data: 'Test'"`
The message "Test" should be written in the logger widget of you VIZ tab.

Thank you to @nicostudt for the link to the FAQ page. The instructions there resolved my issue.

Peppe21, 2022-05-24:
Solved ! 
Many thanks, it is the process doing the publish who was not having the environment variables set. 
Many thanks for the detailed steps Diiliev !

closed 2022-05-24 by Peppe21