ROS-Mobile/ROS-Mobile-Android #91: Could not read the battery state topic using battery widget

amjack0, 2022-06-14:
Hi,

I am using the ROS Mobile to display the battery data on the android tablet. I am publishing the battery data on the battery topic `/battery` and I can see the topic is getting published. I can select my `/battery` topic from the battery widget as well.

But I am unable to see any data on battery GUI. And I get the following error. 

`[ERROR] [1655207807.911237185]: Client [/battery] wants topic /battery to have datatype/md5sum [sensor_msgs/BatteryState/476f837fa6771f6e16e3bf4ef96f8770], but our version has [sensor_msgs/BatteryState/4ddae7f048e32fda22cac764685e3974]. Dropping connection.`

Can anyone please tell me what can be missing here. 

```     
          ros::Publisher battery_pub = nh.advertise<sensor_msgs::BatteryState>("battery", 10);

          batt_msgs.header.stamp = ros::Time::now();
          batt_msgs.present = true;
          batt_msgs.percentage = percentage_cap;
          batt_msgs.voltage = voltage; 
          batt_msgs.current = current;
          batt_msgs.temperature = temperature;

          battery_pub.publish(batt_msgs);

```

System: `ROS Noetic + 20.04 `

nicostudt, 2022-06-15:
Hey, 
it might be possible that the message definitions of our app and your compiled ROS version are different. I'll take a close look and test it again.

nicostudt, 2022-06-15:
We are eagerly waiting for an update of ROSjava to fix most of the problems here ... but unfortunately nothing happens for years.

amjack0, 2022-06-15:
I tried to change the `BatteryState.h` message to fix the error (Although, not recommended but just for testing). The error got fixed, but ROS Mobile app got crashed.

nicostudt, 2022-06-16:
Okay, I took a closer look on it.
We're currently depend on the message definitions of Melodic. In the BatteryState msg (Noetic) there is an additional field "temperature". 

So there are several possibilities now:
1. We have to include a new feature, where you could choose the ROS version in the app. This would be a huge pain thinking about the current implementation of ROSjava.
2. You could change the message definition of BatteryState in Noetic to exclude "temperature".

amjack0, 2022-06-17:
Yes, Thank you for your support. 

I think, It's much better to go with 2nd option. 
with second option do you mean, change the `BatteryState.msg` in `/opt/ros/noetic` or use a Custom msg based on `BatteryState.msg` without the temperature field.

amjack0, 2022-06-17:
Hi,

So, I defined `BatteryState.msg` as a custom `msg` in my package without the `temperature` field (as you recommended). All the errors are gone and the app is also running fine.  

But I am again unable to see any data on battery GUI. 

This is the code for publishing the battery data:

```     
          #include <my_package/BatteryState.h>

          ros::Publisher battery_pub = nh.advertise<my_package::BatteryState>("battery", 10);
          my_package::BatteryState batt_msgs;

          batt_msgs.header.stamp = ros::Time::now();
          batt_msgs.present = true;
          batt_msgs.percentage = percentage_cap;
          batt_msgs.voltage = voltage; 
          batt_msgs.current = current;

          battery_pub.publish(batt_msgs);
``` 

@nicostudt

nicostudt, 2022-06-17:
Mhh, do you use the Play Store version of the app or the custom build from this repo?  My question would be if you have access to the logs and if there is an error issued.

amjack0, 2022-06-18:
I'm using the play Store version of the app.

amjack0, 2022-06-22:
So I tried to change the message definition of `BatteryState` in noetic to exclude the `temperature` field.   
I used the `rqt_graph` to see the node communication. This is what I found,

1. When I use **`sensor_msgs/BatteryState`** to publish the `\battery` topic. The battery node created by ROS Mobile is communicating with the battery topic. But I don't see the data on GUI.
  
2. When I use **`my_package/BatteryState`** to publish the `\battery` topic. The battery node created by ROS Mobile is **NOT** communicating with the battery topic. And of course, I don't see the data on GUI.

nicostudt, 2022-06-23:
I've found a [discussion](https://answers.ros.org/question/363757/how-to-receive-standard-message-that-has-changed-format-in-noetic/) about the changed BatteryState Msg. Apparently it seems that you can republish the BatteryState in the old format, but I'll have to take another look at it.

amjack0, 2022-06-27:
So, I tried the solution suggested in above discussion.
I wrote a node in between the GUI and BatteryState publisher, Which converts `noetic msgs` to `melodic`. It did not help either !
can you provide some help. I am out of options now ;(

Chick92, 2023-05-07:
Having the same issue, any update on this?