ROS-Mobile/ROS-Mobile-Android #92: Possibility to save configurations

maxiew123, 2022-06-16:
Hello, Thanks for building this app. Is it possible for saving my configurations for changing devices. Thank you.

nicostudt, 2022-06-16:
Hey,
in short: not at the moment. 

We are planning to include sharing the config via Firebase or another database or encoded as a QR code or something else. (Do you have some suggestions or requests on how to make this happen?)

maxiew123, 2022-06-17:
Thanks for taking it in consideration. As a QR code does seem like a good idea or maybe via a profile manager.

tolivervix, 2022-08-08:
Hey, this feature will help so much to share profiles with my team. 
Save a profile and import from storage seems to be the best option for my necessity.