ROS-Mobile/ROS-Mobile-Android #94: Visualizing Pointcloud2 message type

RishabhBajaj25, 2022-06-22:
Hi, firstly, thanks for the amazing work. I am using your app to ssh into a robot equipped with lidar and camera to visualise the camera data. Is it possible for the app to also visualise the pointcloud the robot builds using SLAM? If not, is there a workaround for that?

Thanks

nicostudt, 2022-06-23:
Hey,
thank you for using our app!
At the moment Pointclouds are not supported as you said. The only workaround would be to clone the project from Github and create your own 2D Layer Widget for Pointclouds.

Chick92, 2022-08-19:
Fantastic app, are there any plans to officially support 3D pointclouds in the future?

nicostudt, 2022-08-20:
> Fantastic app, are there any plans to officially support 3D pointclouds in the future?

Thank you very much! 
There are plans but it is a complex topic on its own as they would require 3D rendering on the devices. I will look into it, but I can't promise that it will be released in the nearer future.

Chick92, 2022-08-20:
@RishabhBajaj25 I've written a set of instructions to get foxglove studio to work in Google Chrome Beta on an android device. You'll need a mouse and the setup is a bit convoluted, but it's better than nothing. 
https://github.com/foxglove/studio/issues/4107#issuecomment-1220995103