ROS-Mobile/ROS-Mobile-Android #95: Connecting to ROS and give error " HTTP method 'POST' not allowed"

ashish381, 2022-07-04:
2022-07-04 10:44:55.936 785-785/? E/audit: type=1400 audit(1656917095.931:4003): avc:  denied  { kill } for  pid=0 comm="swapper/1" capability=5  scontext=u:r:kernel:s0 tcontext=u:r:kernel:s0 tclass=capability permissive=0 SEPF_SM-A715F_11_0010 audit_filtered
2022-07-04 10:44:56.782 802-853/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2022-07-04 10:44:58.387 802-853/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2022-07-04 10:44:58.387 802-853/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2022-07-04 10:44:58.387 802-853/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2022-07-04 10:44:58.407 13809-20102/com.schneewittchen.rosandroid E/Registrar: Exception caught while communicating with master.
    java.lang.RuntimeException: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:148)
        at java.lang.reflect.Proxy.invoke(Proxy.java:1006)
        at $Proxy2.registerPublisher(Unknown Source)
        at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
        at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
     Caused by: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.checkStatus(XmlRpcCommonsTransport.java:259)
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.getInputStream(XmlRpcCommonsTransport.java:118)
        at org.apache.xmlrpc.client.XmlRpcStreamTransport.sendRequest(XmlRpcStreamTransport.java:152)
        at org.apache.xmlrpc.client.XmlRpcHttpTransport.sendRequest(XmlRpcHttpTransport.java:143)
        at org.apache.xmlrpc.client.XmlRpcClientWorker$1.run(XmlRpcClientWorker.java:80)
        at java.lang.Thread.run(Thread.java:923) 
2022-07-04 10:44:58.407 13809-20103/com.schneewittchen.rosandroid E/Registrar: Exception caught while communicating with master.
    java.lang.RuntimeException: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:148)
        at java.lang.reflect.Proxy.invoke(Proxy.java:1006)
        at $Proxy2.registerPublisher(Unknown Source)
        at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
        at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
     Caused by: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.checkStatus(XmlRpcCommonsTransport.java:259)
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.getInputStream(XmlRpcCommonsTransport.java:118)
        at org.apache.xmlrpc.client.XmlRpcStreamTransport.sendRequest(XmlRpcStreamTransport.java:152)
        at org.apache.xmlrpc.client.XmlRpcHttpTransport.sendRequest(XmlRpcHttpTransport.java:143)
        at org.apache.xmlrpc.client.XmlRpcClientWorker$1.run(XmlRpcClientWorker.java:80)
        at java.lang.Thread.run(Thread.java:923) 
2022-07-04 10:44:58.414 13809-20104/com.schneewittchen.rosandroid E/Registrar: Exception caught while communicating with master.
    java.lang.RuntimeException: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:148)
        at java.lang.reflect.Proxy.invoke(Proxy.java:1006)
        at $Proxy2.registerPublisher(Unknown Source)
        at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
        at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[... 97 lines cut ...]
        at $Proxy2.registerPublisher(Unknown Source)
        at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
        at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
        at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
        at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
     Caused by: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.checkStatus(XmlRpcCommonsTransport.java:259)
        at org.apache.xmlrpc.client.XmlRpcCommonsTransport.getInputStream(XmlRpcCommonsTransport.java:118)
        at org.apache.xmlrpc.client.XmlRpcStreamTransport.sendRequest(XmlRpcStreamTransport.java:152)
        at org.apache.xmlrpc.client.XmlRpcHttpTransport.sendRequest(XmlRpcHttpTransport.java:143)
        at org.apache.xmlrpc.client.XmlRpcClientWorker$1.run(XmlRpcClientWorker.java:80)
        at java.lang.Thread.run(Thread.java:923) 
2022-07-04 10:45:08.487 13809-20103/com.schneewittchen.rosandroid E/Registrar: Exception caught while communicating with master.
    java.lang.RuntimeException: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed
        at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:148)
        at java.lang.reflect.Proxy.invoke(Proxy.java:1006)

nicostudt, 2022-07-04:
Hey,
Can you provide some more information like: 
What version of the app are you using?
What are your settings or used widgets during the error?
Are you using WIFI or LTE?

Thanks in advance!

ashish381, 2022-07-05:
> 

Hi nicostudt, Thanks for your response 
 Please go through following points:
1. I am using Android 10 version 
2. First I am trying to connect with ip and port and it showing "Connected" but giving me response  "java.lang.RuntimeException: org.apache.xmlrpc.client.XmlRpcHttpTransportException: HTTP server returned unexpected status: HTTP method 'POST' not allowed".
3. In SSH connection , I am not able to connect , I can't see any terminal response
4. I am on  WIFI in same network,

ashish381, 2022-07-05:
Hi nicostudt, my error solved , i was giving wrong port on master fragment

closed 2022-07-05 by ashish381