ROS-Mobile/ROS-Mobile-Android #97: How to use realsense as camera, and display in ROS-Mobile

28598519a, 2022-09-13:
I use NVIDIA Jetson TX2 on my turtlebot3 and have Intel realsense D435i installed

How to use realsense as camera, and display in ROS-Mobile?

Also, is there a way to show point cloud in the app? like I'm running it directly from the Turtlebot3 terminal?

For eample:
`roslaunch realsense2_camera demo_pointcloud.launch`
or
`roslaunch turtlebot3_slam turtlebot3_slam.launch`

nicostudt, 2022-09-14:
Hey,

follow the instruction when connecting to your ROS master to make sure that the app can successfully communicate with your ROS nodes. If the connection fails, a popup with a checklist should open.

In the Details tab add a Camera widget and edit the topic to a published topic of your camera.
I highly recommend to subscibe to ".../compressed" topics as otherwise you will get a lot of frame loss due to network bandwidth.

Good luck!

closed 2022-11-09 by nicostudt

punlerleng706-cmd, 2026-11-08:
@article{rottmann2020ros,
  title={ROS-Mobile: An Android application for the Robot Operating System},
  author={Rottmann, Nils and Studt, Nico and Ernst, Floris and Rueckert, Elmar},
  journal={arXiv preprint arXiv:2011.02781},
  year={2020}
}