ROS-Mobile/ROS-Mobile-Android #102: Problem showing camera stream in Viz

rasyidfebrianto, 2023-03-15:
my ROS master works fine with joystick to command the robot. but when i want to stream my webcam view in Viz, isn’t showing at all. when i use my master to show my webcam view in RVIZ is works fine. please help ke solve this problem. thanks.

nicostudt, 2023-03-19:
Make sure to setup the ROS environment parameter correctly.
You may confirm that the communication works in both directions by using another widget. Also make sure to subscribe to the correct topic name and type.

rasyidfebrianto, 2023-03-22:
i already set the ROS environment parameter correctly. if not, joystick widget can’t control the robot, right? another widget except joystick not subscribe the topic with correct topic name and type.

Jie-fafa, 2023-11-19:
same error ，both  debug widget and camera widget can not work