ROS-Mobile/ROS-Mobile-Android #106: Debug Widget

Imaaddii, 2023-04-24:
Hi,

I can connect to my ROS master and see all topics, added debug Widget and select /cmd_vel from dropdown menu. everything working fine, except there is no data shown in VIZ 
On the other hand in if i select joystick and publish data on /cmd_vel then its publish on my turtle_bot node and working fine

Am I missing something in my android settings, how to fix that it also subscribe data from subscribe data from active topics  ?

SamueleSandrini, 2023-08-28:
Same issue for me; as far as widgets that publish something are concerned, they work perfectly (on the ros master running on another pc you can see the published messages), vice versa for widgets working from subscribers they don't work and I don't see the posts in viz. 
Did you solve it somehow?

Jie-fafa, 2023-11-19:
same error