ROS-Mobile/ROS-Mobile-Android #111: Nodes do not subscribe the topics

oguzhanbzglu, 2023-10-02:
I correctly connected to my computer through SSH and can run the scripts or launch files. Unfortunately, when I run the Joystick widget in Viz section it does not work there.

mxpfeng, 2023-10-26:
I have the same problem as you. Has anyone solved it？

oguzhanbzglu, 2023-10-26:
I have no idea. I thought it's because of I use different ROS destro and version. But, I still run the commands on terminal screen on my phone.

mxpfeng, 2024-07-19:
> 我不知道。我以为是因为我使用了不同的 ROS destro 和版本。但是，我仍然在手机的终端屏幕上运行命令。

I abandoned this project before and used another one instead. Recently, I started using this project again and found a solution to the problem. As described in 'https://github.com/ROS-Mobile/ROS-Mobile-Android/wiki/FAQ', you need to add the Android IP to the host. ROS_IP='IP address of the Android device'.

mxpfeng, 2024-07-19:
> I have no idea. I thought it's because of I use different ROS destro and version. But, I still run the commands on terminal screen on my phone.

I abandoned this project before and used another one instead. Recently, I started using this project again and found a solution to the problem. As described in 'https://github.com/ROS-Mobile/ROS-Mobile-Android/wiki/FAQ', you need to add the Android IP to the host. ROS_IP='IP address of the Android device'.